BehaviorTree/BehaviorTree.CPP #939: Test failures when building with clang

dsobek, 2025-02-25:
**Describe the bug**
A clear and concise description of what the bug is.

When building with clang on master, tests are failing with c++ exception:
`C++ exception with description "Value outside numeric limits" thrown in the test body.`

`Any.Cast` appears to be the most minimal testcase that is failing. It comes down to casts from any integral type to a double type.

**How to Reproduce**

Please provide a specific description of how to reproduce the issue or source code that can be compiled and executed. Please attach a file/project that is easy to compile, don't copy and paste code snippets!

From project root:
```
CC=clang CXX=clang++ colcon build
./build/behaviortree_cpp/tests/behaviortree_cpp_test
```

**Root cause**

I've found that the root cause is in [checkTruncation](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/a7c90872ee776c840044e49e260ecde69bd69fab/include/behaviortree_cpp/utils/convert_impl.hpp#L90)

In particular if `To` has a larger max value than `From` (say for example `To` is `double` and `From` is `int`),  `static_cast<From>(std::numeric_limits<To>::max())` does not always evaluate to the max value of `From`. In clang, with any optimization (I see behaviortree.cpp is built with `-O3`), this evaluates to 0. I am not sure if this is meant to be undefined behavior.

dsobek, 2025-02-25:
Potential solution in https://github.com/BehaviorTree/BehaviorTree.CPP/pull/940

Aglargil, 2025-02-25:
This might help too #932

facontidavide, 2025-02-25:
I am aware of the issue.

facontidavide, 2025-02-25:
the change in the code was introduced mased on #919 , as suggested by @cktii

but clearly that is not a solution that works on all platforms.

@Aglargil how is #932 related?

@dsobek your PR #940 doesn't pass the tests in wondows

Aglargil, 2025-02-25:
> [@Aglargil](https://github.com/Aglargil) how is [#932](https://github.com/BehaviorTree/BehaviorTree.CPP/pull/932) related?

@facontidavide 
I found this error when I was adding the _onStart precondition function to the script, so I fixed it, you can take a look at this commit: https://github.com/BehaviorTree/BehaviorTree.CPP/commit/d8ca1f1b055aaf50d4c95b26f204e787d77b6b71

facontidavide, 2025-02-25:
fixed with #941 .Thanks @Aglargil you were right!

closed 2025-02-25 by facontidavide

dsobek, 2025-02-25:
Thanks for fixing!

> [@dsobek](https://github.com/dsobek) your PR [#940](https://github.com/BehaviorTree/BehaviorTree.CPP/pull/940) doesn't pass the tests in wondows

For the record, I think you were looking at Picknik's fork in https://github.com/PickNikRobotics/BehaviorTree.CPP/pull/6 where we haven't gotten around to fixing the windows build because we don't support it. I don't think https://github.com/BehaviorTree/BehaviorTree.CPP/pull/940 would've shown the same failures.