BehaviorTree/BehaviorTree.CPP #943: Allow passing a derived object to a port requiring a base class

captain-yoshi, 2025-03-01:
**Is your feature request related to a problem? Please describe.**
There is a lot of boilerplate code needed for converting libraries, relying on inheritance, into BT nodes. Consider this example:
```cpp
// Base class
class Greeter {
 public:
  virtual std::string show_msg() const = 0;
  virtual ~Greeter() {}
};

// Derived class
class HelloGreeter : public Greeter {
 public:
  std::string show_msg() const override { return "hello"; }
  void setDerivedParameter(int n);
};
```
I have 2 options if I want to use a unique method from the derived class as well as store it's base class somewhere, e.g. inside a vector.
```xml
<!-- OPTION 1- return base and derived class -->
<CreateHelloGreeter out_derived="{hello_greeter}" out_base="{greeter}" />

<!-- OPTION 2- convert to base -->
<CreateHelloGreeter out_derived="{hello_greeter}" />
<UpcastHelloGreeter in={hello_greater} out_base="{greeter}" />

<setDerivedParameter in_derived="{hello_greater}" n="2" />
<VectorInsertGreeter in_base="{greeter}" />
```

**Describe the solution you'd like**
I would like the ability to allow passing a derived object directly to a port requiring a base class.
```xml
<CreateHelloGreeter out_derived="{hello_greeter}" />

<setDerivedParameter in_derived="{hello_greater}" n="2" />
<InsertToGreeterVector in_base="{hello_greeter}" />
```
As you see, this would reduce boilerplate code in the C++ and in the XML representation. Is there any way to reduce the strictness of the custom `std::any` used in this project to enable this feature ? I think that this project https://github.com/samee/any_of implements this behavior.

There is also the problem that BT evaluates ports type for consistency. I don't think one can deduce the base class of a derived class 
 `typeid`...

facontidavide, 2025-03-01:
this is a good point.

Yes, I think it is safe to reduce the strictness of the any_cast.

Can you please help creating a PR with a unit test of the desired behavior, i.e. somehting like this should work out of the box

```xml
<CreateHelloGreeter out_derived="{hello_greeter}" />
<setDerivedParameter in_derived="{hello_greater}" n="2" />
<ShowGreetmessage in_base="{hello_greeter}" />
```

facontidavide, 2026-10-24:
Fixed by #1107 (1e166f51). Register the inheritance once with `factory.registerPolymorphicCast<Derived, Base>()`, and a `std::shared_ptr<Derived>` output can then be read by a `std::shared_ptr<Base>` input port (see `tests/gtest_polymorphic_ports.cpp`). Making the cast automatic, without registration, would need a separate feature request.

closed 2026-10-24 by facontidavide