BehaviorTree/BehaviorTree.CPP #957: Adding default ports to subtrees

ufrhaidar, 2025-04-08:
**Is your feature request related to a problem? Please describe.**
Want to use subtrees more often to reduce repeated logic and to better explain sections of code (clean/clear up trees and reduce repeated logic), but in most cases (depending on how many ports the nodes i want to place in a subtree use), it takes longer to create the port mappings than to just leave the logic as is or create messy trees.

**Describe the solution you'd like**
Having the ability to add default ports and values for subtrees. or another button similar to `_autoremap` that creates ports for all black bord variables used in subtree?

**Describe alternatives you've considered**
copy pasting prefilled tree. you need to know where it was used. not easy in more complicated trees.

**Additional context**
Something similar to behaviors/containers in flexbe.

facontidavide, 2025-04-17:
this is already possible, checj this example

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/examples/t14_subtree_model.cpp

closed 2025-04-17 by facontidavide

ufrhaidar, 2025-05-01:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<root BTCPP_format="4">
  <BehaviorTree ID="ExampleSubtree">
    <Script code="output=input"/>
  </BehaviorTree>

  <!-- Description of Node Models (used by Groot) -->
  <TreeNodesModel/>

</root>
```

If I was to manually implement your example above in the subtree xml created by groot:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<root BTCPP_format="4">
  <BehaviorTree ID="ExampleSubtree">
    <Script code="output=input"/>
  </BehaviorTree>

  <!-- Description of Node Models (used by Groot) -->
  <TreeNodesModel>
    <SubTree ID="ExampleSubtree">
      <input_port name="input"/>
      <output_port name="output"/>
    </SubTree>
  </TreeNodesModel>

</root>
```
- if I load updated xml, the `TreeNodesModel` can be used as intended, but if i was to resave the subtree, the modified `TreeNodesModel` gets deleted by groot unless it is used by something else in that same xml file. (i assume groot deletes unused nodes from `TreeNodesModel`)

Am I using it wrong? I tend to have subtrees that can be shared by different projects saved in a separate xml, so I wouldn't always use the subtree in the same xml. 

to get around this, I am adding an unused subtree that uses my subtree
```xml
<?xml version="1.0" encoding="UTF-8"?>
<root BTCPP_format="4">
  <BehaviorTree ID="ExampleSubtree">
    <Script code="output=input"/>
  </BehaviorTree>

  <BehaviorTree ID="_">
    <SubTree ID="ExampleSubtree"
             input=""
             output=""/>
  </BehaviorTree>

  <!-- Description of Node Models (used by Groot) -->
  <TreeNodesModel>
    <SubTree ID="ExampleSubtree">
      <input_port name="input"/>
      <output_port name="output"/>
    </SubTree>
  </TreeNodesModel>

</root>
```