BehaviorTree/BehaviorTree.CPP #976: Package isn't available anymore under ROS2 rolling

Darkproduct, 2025-05-20:
**Describe the bug**
Since last week, the package `ros-rolling-behaviortree-cpp` isn't available anymore. I know it was available previously and [from the ROS index](https://index.ros.org/p/behaviortree_cpp/#rolling) it looks like as if it was still available, but on `apt` *thinks* otherwise.

*How to Reproduce**
Try installing `ros-rolling-behaviortree-cpp` with `sudo apt install ros-rolling-behaviortree-cpp`.

SteveMacenski, 2025-05-28:
I believe that this has been resolved if you do an apt update

SteveMacenski, 2025-05-28:
Any update when this would be available again?

facontidavide, 2025-05-29:
[image]

hope this helps

SteveMacenski, 2025-05-29:
I'm not entirely sure why your CI didn't pick it up, I haven't used the industrial_ci from ros industrial. I spent ~10 minutes going through it and I'm not finding anything immediately obvious to me wrong. Maybe its worth setting up the ROS build farm's `test_pull_requests` to true so that the build farm does this for you to see what it sees. You have to setup a web hook in the project's setting so the job automatically pops up in the PR github workflows UX, but that's a one-time set it and forget it thing.