BehaviorTree/BehaviorTree.CPP #985: Problem in compiling with conan

tahayf, 2025-06-20:
When I try to compile the project with conan, at last step in "cmake --build . --parallel" I get an error like this "C:\Users\PC_6334\Desktop\behavior\BehaviorTree.CPP\3rdparty\cpp-sqlite\sqlite.hpp(24,10): error C1083: cannot open include file: 'sqlite3.h': No such file or directory [C:\Users\PC_6334\Desktop\behavior\bui 
ld\behaviortree_cpp.vcxproj]"

How can I fix this? Is it something wrong with my system? Conan install results say that I'm successfully installed sqlite.

tahayf, 2025-07-14:
dude c'mon compiling at the beginning with conan is totally broken even if I do all steps right it still throws bunch of errors in windows mostly related to dependencies "sqlite3" and "zeromq"

facontidavide, 2025-07-14:
As you can appreciate in both the Ubuntu and Windows worklow files here, Conan is working just fine. Please compare what you are doing with those steps, to see if you are doing anything different.

commit 4d6f92ad (2025-07-15): Update README.md fix #985

Duuuude