BehaviorTree/BehaviorTree.CPP #998: Porting useful ControlNodes from Nav2 as stand-alone nodes in BehaviorTree.CPP

Mechazo11, 2025-08-24:
Hi @facontidavide 

I am a robotics researcher from USA and in a recent internship, we have been extensively using BehaviorTree.CPP v3.8 for orchestrate a variety of autonomy and robot navigation related tasks. This backdrop leads to the following feature request

## Feature Proposal 
Nav2, a popular open-source project from OpenNavigation is known for adding a number of useful Control, Action and Decorator nodes. Some of the most useful ones are `PipelineSequence`, `RecoveryNode` and `RateController`

Some autonomy related projects may have constraints of not using ROS 2 ecosystem. Without ROS 2 and by extension Nav2, these useful nodes become inaccessible natively from BehaviorTree.CPP.

## Proposed Solution

I propose to open a PR for `v3.8` branch that backports a number Nav2 nodes as native BehaviorTree.CPP nodes. 

From what I have found out so far, the implementation of these nodes may have minimal dependency on the actual Nav2 stack. An example:  [pipeline_sequence.hpp](https://github.com/ros-navigation/navigation2/blob/69a60dfacb80eb3c9c2ad47f7590eea8defbdaec/nav2_behavior_tree/include/nav2_behavior_tree/plugins/control/pipeline_sequence.hpp#L55)

In my opinion, the following nodes may be most suitable for application across various domains
 
### Control nodes
  * PipelineSequence: reticks preceding child nodes that return `SUCCESS`. 
  * RecoveryNode: Two child node that only returns `SUCCESS` if and only if left child is successful.  
  * NoneblockingSequence: NonblockingSequence ticks all children until one fails, re-ticking running/successful ones, and returns SUCCESS only if all succeed.
 
### Decorator nodes
  * Rate Controller:  "Controls the ticking of its child node at a constant frequency. The tick rate is an exposed port"

## Additional Context

Why `v3.8`? I have recently worked with a number of robots from notable robotics companies who are still using ROS 2 Humble and the humble branch of Nav2. As far as I know, BehaviorTree.CPP v3.8 is the version supported by Nav2 Humble.

Mechazo11, 2025-08-27:
Hi @SteveMacenski thank you for your insight. From my current internship experience, the approach I have seen is that `autonomy` and `navigation` are regarded as two separate modules. All autonomous robot behaviors (re-planning, collision avoidance, point-to-point navigation etc.) are regarded as methods/skills within `navigation` module whilst  `autonomy` is regarded more as an [Agentic AI](https://blogs.nvidia.com/blog/what-is-agentic-ai/) that acquires `navigation` along with other modules to perform various tasks defined in human language.

However, a number of Nav2's BT nodes appears to be quite useful to orchestrate the `autonomy`'s stack itself but sometimes these trees needs to be Nav2 independent (in one case it was even ROS 2 agnostic). This is what lead me to this proposal.

facontidavide, 2026-10-24:
Closing as stale: these nodes live in Nav2, and v3.8 isn't receiving feature backports. As said earlier, a PR adding them to master (with docs, SKIPPED handling and tests) is still welcome if someone wants to carry it.

closed 2026-10-24 by facontidavide